gazebosim/ros_gz #701: Adopt ros_gz_bridge format for ros_gz_image

TannerGilbert, 2025-02-28:
Currently the `ros_gz_image` bridge only takes Gazebo image topics as arguments and doesn't allow them to be renammed unlike the `ros_gz_bridge`. I think it would be more intuitive if both bridges had the same input format and also both support specifying the topics via arguments and via config file.

Example the following should also work for the image bridge (in my opinion):
```
- ros_topic_name: "navigation_camera/image"
  gz_topic_name: "/world/map/model/iris/link/navigation_camera_link/sensor/navigation_camera/image"
  ros_type_name: "sensor_msgs/msg/Image"
  gz_type_name: "gz.msgs.Image"
  direction: GZ_TO_ROS
```

Additionally I would like for the `ros_gz_image` bridge to also bridge the `camera_info` topic as for me it also belongs to the camera and it's not really intuitive to have to bridge it separately with the `ros_gz_bridge`.

TZECHIN6, 2025-04-22:
Push, just come across this... look up the document and fail to change the topic name in ROS.

jaiminlabs, 2026-10-11:
Hi, I'd like to work on this. Could I be assigned?

I'd like to start with the first part: making ros_gz_image accept ros_topic_name and gz_topic_name, plus config file support, to match ros_gz_bridge. I'd suggest handling camera_info as a separate follow-up, since it may need its own design discussion. Does this split work for you?

oKermorgant, 2026-10-26:
Hi, just came to this when following the rabbit hole after getting `optical_frame_id deprecated` messages.
`ros_gz_image` should indeed be compatible with a YAML file, and also the `frame_id` change so that multiple bridges can run in a single executable *and* update the frames, which is not feasible with a single `override_frame_id` parameter.
I agree that `camera_info` is another issue, as you cannot always derive the `CameraInfo` topic from the image topic so you need additional topic names just in case. Yet, it would make sense to regroup the 2 of them in this pkg.