gazebosim/ros_gz #708: Improve documentation

ahcorde, 2025-03-18:
We have two diferrent places where the documentation is written:

 - In the `README.md` of each repository
   - [ros_gz_bridge](https://github.com/gazebosim/ros_gz/tree/ros2/ros_gz_bridge)
   - [ros_gz_image](https://github.com/gazebosim/ros_gz/tree/ros2/ros_gz_image)
   - [ros_gz_interfaces](https://github.com/gazebosim/ros_gz/tree/ros2/ros_gz_interfaces)
   - [ros_gz_sim](https://github.com/gazebosim/ros_gz/tree/ros2/ros_gz_sim)
   - [ros_gz_sim_demos](https://github.com/gazebosim/ros_gz/tree/ros2/ros_gz_sim_demos)
 - In the official Gazebo documentation [1]

The idea is to move all the documentation to the Gazebosim docs website [2]

[1] https://gazebosim.org/docs/latest/ros2_overview/ 
[2] https://github.com/gazebosim/docs

FYI @j-rivero

j-rivero, 2025-03-18:
Thanks for starting the task! Some feedback:

> The idea is to move all the documentation to the Gazebosim docs website [2]

+1. What we can do is to leave pointers from the READMEs to the official documentation so people can quickly find the right links.

Reviewing the documentation in https://gazebosim.org/docs/latest/tutorials/:

[image]

We probably need to be a bit more descriptive in our tittles so people can know what to expect when opening the link. For example: [ROS 2 Integration Template](https://gazebosim.org/docs/latest/ros_gz_project_template_guide/) could be renamed to something like "ROS 2 and Gazebo repository template and workflow". 

We can also add subsections here to organize the tutorials if we migrate a good bunch of them and the list grows a lot. I'm missing here at least a link to the ROS/Gazebo installation (although it is in the index) and one to the high level overview https://gazebosim.org/docs/latest/ros2_overview/.

Now looking into the Index entries:

[image]

Would be great if all the entries start with the same prefix: "ROS 2: launching Gazebo", "ROS 2: interacting with Gazebo",  etc.

LeoLin6, 2025-03-19:
Could I give this a try? Just to make sure, we would like the README files to just be links to the documentation?

j-rivero, 2025-03-20:
> Could I give this a try? 

Sure, thanks! The main documentation repository is https://github.com/gazebosim/docs . I think that the changes can land in of the active releases (Ionic or Harmonic probably) and after that we can copy the changes to the other releases in that repository.

> Just to make sure, we would like the README files to just be links to the documentation?

That is the idea, the README file can contain a bit of text explaining that the relevant documentation can be found in the links to the documentation site.

Amronos, 2025-03-26:
My comment [here](https://github.com/gazebosim/ros_gz/issues/716#issuecomment-2754941571) can also probably be edited and added to the documentation.

dvdmc, 2025-03-26:
Joining! @LeoLin6 how is your process? Are you currently working on it? I will leave some feedback here and work on it when I have some time in the next days unless you are already on it.

@j-rivero adding to your comments: Since all ROS-related tutorials will be prepended by ROS 2, probably it makes sense to Create a ROS 2 sublist. Maybe even on the top level so ROS users have a clear route. These tutorials could start with the ROS/Gazebo installation. Then move on incrementally in usability and complexity. This is based on my ROS user experience and teacher which I don't think as extensive, but the order a user could follow:
- Launch a simulation and spawn a robot with communication (bridge) to ROS.
- Introduce this process within their method (launch simulation and bridge from ros)
- Leading to more stable integration
- Need for composition and other lower-level knowledge

Based on this I reasoned on the tutorial ordering. This is intended to be on a section level but I would revise the texts for each section:
 
- Spawning a URDF is correctly placed but the approach is not related to ROS. The ROS version of this tutorial is currently in "Spawn a Gazebo model from ROS 2" which is well below the beginning of ROS tutorials. I would include here standard practices on how spawning should be done. There is no example of a Python launch file.
- ROS integration overview is good and explanatory but I think including composition here might be misleading and would push it for later. I would also include here a review of terminology for `gz sim` and `gz_ros` concepts. Probably also include the contents from "Use ROS2 to interact with Gazebo" regarding the bridge (This is currently after the "Launch..." tutorial).
- Launch Gazebo from ROS is well placed. It currently lacks quite some content as outlined in #716. I would include **standard practices** and examples about how to run and include all the components (gz_server, gz_client, gz_ros_bridge) in XML/Python launchfiles similar to "Use ROS2 to interact with Gazebo".
- The integration template tutorial is well placed. I have a question regarding the template below.
- I think the "ROS 2 interoperability" tutorial objective is not clear. What's its specific purpose or topic?

Any thoughts on this structure? Anything missing?

Question regarding integration template: `bringup`, `description`, and `gazebo` are related to specific robots or platforms. In research groups they could potentially include their specific robots, sensors... However, in this tutorial they are associated with a "Project". I would say that in research groups, `bringup`, `description`, and `gazebo` are usually dependencies spread across other repositories external to the project. Would it make sense to structure the template tutorial around the concept of the these three folders and treat them independently from a "Project"? Then, focus instead on how they should be added as "external" dependencies to any project.

j-rivero, 2025-03-27:
> Joining! @LeoLin6 <https://github.com/LeoLin6> how is your process? Are
> you currently working on it? I will leave some feedback here and work on it
> when I have some time in the next days unless you are already on it.
>
> @j-rivero <https://github.com/j-rivero> adding to your comments: Since
> all ROS-related tutorials will be prepended by ROS 2, probably it makes
> sense to Create a ROS 2 sublist. Maybe even on the top level so ROS users
> have a clear route. These tutorials could start with the ROS/Gazebo
> installation. Then move on incrementally in usability and complexity. This
> is based on my ROS user experience and teacher which I don't think as
> extensive, but the order a user could follow:
>
>    - Launch a simulation and spawn a robot with communication (bridge) to
>    ROS.
>    - Introduce this process within their method (launch simulation and
>    bridge from ros)
>    - Leading to more stable integration
>    - Need for composition and other lower-level knowledge
>
> Based on this I reasoned on the tutorial ordering. This is intended to be
> on a section level but I would revise the texts for each section:
>
>    - Spawning a URDF is correctly placed but the approach is not related
>    to ROS. The ROS version of this tutorial is currently in "Spawn a Gazebo
>    model from ROS 2" which is well below the beginning of ROS tutorials. I
>    would include here standard practices on how spawning should be done. There
>    is no example of a Python launch file.
>    - ROS integration overview is good and explanatory but I think
>    including composition here might be misleading and would push it for later.
>    I would also include here a review of terminology for gz sim and gz_ros
>    concepts. Probably also include the contents from "Use ROS2 to interact
>    with Gazebo" regarding the bridge (This is currently after the "Launch..."
>    tutorial).
>    - Launch Gazebo from ROS is well placed. It currently lacks quite some
>    content as outlined in #716
>    <https://github.com/gazebosim/ros_gz/issues/716>. I would include *standard
>    practices* and examples about how to run and include all the
>    components (gz_server, gz_client, gz_ros_bridge) in XML/Python launchfiles
>    similar to "Use ROS2 to interact with Gazebo".
>    - The integration template tutorial is well placed. I have a question
>    regarding the template below.
>    - I think the "ROS 2 interoperability" tutorial objective is not
>    clear. What's its specific purpose or topic?
>
> Any thoughts on this structure? Anything missing?
>
> Question regarding integration template: bringup, description, and gazebo
> are related to specific robots or platforms. In research groups they could
> potentially include their specific robots, sensors... However, in this
> tutorial they are associated with a "Project". I would say that in research
> groups, bringup, description, and gazebo are usually dependencies spread
> across other repositories external to the project. Would it make sense to
> structure the template tutorial around the concept of the these three
> folders and treat them independently from a "Project"? Then, focus instead
> on how they should be added as "external" dependencies to any project.
>
>
> @j-rivero <https://github.com/j-rivero> adding to your comments: Since
> all ROS-related tutorials will be prepended by ROS 2, probably it makes
> sense to Create a ROS 2 sublist. Maybe even on the top level so ROS users
> have a clear route.
Thanks David for the comments. With respect to how and where to place them
I think that it makes sense for them to be under 'tutorials'. The option of
creating a sublist makes sense to me if it looks good with the multilevel
approach in the menu.

These tutorials could start with the ROS/Gazebo installation. Then move on
> incrementally in usability and complexity.
+1

> This is based on my ROS user experience and teacher which I don't think as
> extensive, but the order a user could follow:
>
>    - Launch a simulation and spawn a robot with communication (bridge) to [...]
That looks correct to me, thanks for putting it together. We need to see
how much we migrate from this repository that is currently I'm the readmes
and fit into the structure.

First PRs can be to modify the index to a multilevel approach and a second
one to re-organize what it is In the docs currently.

> Question regarding integration template: bringup, description, and gazebo
> are related to specific robots or platforms. In research groups they could
> potentially include their specific robots, [...]
We can re-think this indeed. Could you please open an issue in the template
repository so we discuss it there?

Thanks so much David, really appreciated!

j-rivero, 2025-03-27:
> [@j-rivero](https://github.com/j-rivero) adding to your comments: Since all ROS-related tutorials will be prepended by ROS 2, probably it makes sense to Create a ROS 2 sublist. Maybe even on the top level so ROS users have a clear route.

Thanks David for the comments. With respect to how and where to place them I think that it makes sense for them to be under 'tutorials'. The option of creating a sublist makes sense to me if it looks good with the multilevel approach in the menu.

>  These tutorials could start with the ROS/Gazebo installation. Then move on incrementally in usability and complexity. 

+1

> This is based on my ROS user experience and teacher which I don't think as extensive, but the order a user could follow:
>
>     * Launch a simulation and spawn a robot with communication (bridge) to [...]
...
> Any thoughts on this structure? Anything missing?

That looks correct to me, thanks for putting it together. We need to see how much we migrate from this repository that is currently I'm the readmes and fit into the structure.

First PRs can be to modify the index to a multilevel approach and a second one to re-organize what it is In the docs currently.

> Question regarding integration template: `bringup`, `description`, and `gazebo` are related to specific robots or platforms. In research groups they could potentially include their specific robots, [...]

We can re-think this indeed. Could you please open an issue in the template repository so we discuss it there?

Thanks so much David, really appreciated!

dvdmc, 2025-03-28:
> Thanks David for the comments. With respect to how and where to place them I think that it makes sense for them to be under 'tutorials'. The option of creating a sublist makes sense to me if it looks good with the multilevel approach in the menu.

Just to confirm: Tutorials -> ROS -> [list] rather than having Tutorial and ROS tutorials->[list] on the same level (?)

> That looks correct to me, thanks for putting it together. We need to see how much we migrate from this repository that is currently I'm the readmes and fit into the structure.

True! My list didn't cover any of the info from this repo. I will try to integrate all of this content into my outline creating new sections if necessary. I will get back here regarding this while we work on the first steps.

> First PRs can be to modify the index to a multilevel approach and a second one to re-organize what it is In the docs currently.

Will work on it. It seems that multi-level is not possible at the moment (see https://github.com/gazebosim/docs/blob/a574980d6aa853ac52af0a03f02f2e08d345fa89/build_multiversion.py#L127) and I'm having some problems building the docs. I will open issues in the docs repo.

> We can re-think this indeed. Could you please open an issue in the template repository so we discuss it there?

Open on https://github.com/gazebosim/docs/issues/580

j-rivero, 2025-03-28:
> Just to confirm: Tutorials -> ROS -> [list] rather than having Tutorial and ROS tutorials->[list] on the same level (?)

Yes, that is my take on it. Not sure how is going to be the look in this two level layout. If it is awful we can revert to use the idea of 'Tutorials' and 'ROS Tutorials".

> True! My list didn't cover any of the info from this repo. I will try to integrate all of this content into my outline creating new sections if necessary. I will get back here regarding this while [...]

Amazing thanks !

> > First PRs can be to modify the index to a multilevel approach and a second one to re-organize what it is In the docs currently.
> 
> Waiting to prepare a pull request from https://github.com/dvdmc/docs/tree/reorg-ros2-tutorials to the **Harmonic docs only** as [@j-rivero](https://github.com/j-rivero) mentioned in his previous comment. There are two problems to discuss for our approach:
> 
>     * It seems that multi-level is not possible at the moment (see https://github.com/gazebosim/docs/blob/a574980d6aa853ac52af0a03f02f2e08d345fa89/build_multiversion.py#L127) so I added it alongside the Tutorials, which will also be more visible to ROS users.
> 
>     * ROS/Gazebo install is applied to all the docs, so it cannot be moved to the ROS 2 Tutorials unless it's done for all the versions.
> 
> I moved this discussion to [gazebosim/docs#582](https://github.com/gazebosim/docs/issues/582)
>

Thanks, let's follow there.
 
> > We can re-think this indeed. Could you please open an issue in the template repository so we discuss it there?
> 
> Open on [gazebosim/docs#580](https://github.com/gazebosim/docs/issues/580)
> 
> The conversation on this issue should probably be only about how to transfer the READMEs to docs from now :)

Thanks, I'll give it a look too.

kscottz, 2025-04-04:
@LeoLin6 Just checking in, how you doing?

LeoLin6, 2025-04-04:
Hey, sorry for missing the other messags, I've honestly been busy so I haven't made too much progress, if yall could contribute it would be great

LeoLin6, 2025-04-04:
@dvdmc Hey David I haven't been able to find time recently to make too much progress, you can definitely go ahead and contribute if you can, thanks

dvdmc, 2025-04-05:
Hey @LeoLin6, kind of the same. Just as a short TL;DR: We moved the main "improve documentation" [here](https://github.com/gazebosim/docs/issues/582). The issue in this repository will cover moving the documentation from the READMEs to the docs repo as you discussed before.