gazebosim/ros_gz #711: SpawnEntity, DeleteEntity, and SetEntityPose NOT working with ros_gz_bridge

rafiqrana, 2025-03-20:
I'm trying to bridge specific Gazebo services to ROS2 using the `ros_gz_bridge` package, but I'm having trouble getting the services to appear in ROS2.

**Environment:**
- ROS2 Iron
- Gazebo Fortress

**Observed Issues:**
 - Services don't appear in ros2 service list
 - No visible errors in bridge node output

**What I've Tried:**
- Verified Gazebo services exist using `gz service -l`

```
rana@rock:~/colcon_ws$ ign service -l
/world/empty/control
/world/empty/control/state
/world/empty/create
/world/empty/create_multiple
/world/empty/declare_parameter
/world/empty/disable_collision
/world/empty/enable_collision
/world/empty/entity/system/add
/world/empty/generate_world_sdf
/world/empty/get_parameter
/world/empty/gui/info
/world/empty/level/set_performer
/world/empty/light_config
/world/empty/list_parameters
/world/empty/playback/control
/world/empty/remove
/world/empty/scene/graph
/world/empty/scene/info
/world/empty/set_parameter
/world/empty/set_physics
/world/empty/set_pose
/world/empty/set_pose_vector
/world/empty/set_spherical_coordinates
/world/empty/state
/world/empty/state_async
/world/empty/system/info
/world/empty/visual_config
/world/empty/wheel_slip

```

- Running the node using the following configuration in my launch file:

```
Node(
    package='ros_gz_bridge',
    executable='parameter_bridge',
    arguments=[
        '/clock' + '@rosgraph_msgs/msg/Clock' + '[gz.msgs.Clock',
        f'/world/empty/control@ros_gz_interfaces/srv/ControlWorld',
        f'/world/empty/create@ros_gz_interfaces/srv/SpawnEntity',
        f'/world/empty/remove@ros_gz_interfaces/srv/DeleteEntity',
        f'/world/empty/set_pose@ros_gz_interfaces/srv/SetEntityPose',
    ],
    output='screen',
),
```

**Expected Behavior:**
- `ros2 service list` should list the following:

```
rana@rock:~/colcon_ws$ros2 service list
/world/empty/control (World control services)
/world/empty/create (Entity spawning)
/world/empty/remove (Entity deletion)
/world/empty/set_pose (Pose updates)
```
- What I get:
```
rana@rock:~/colcon_ws$ros2 service list
/ros_gz_bridge/describe_parameters
/ros_gz_bridge/get_parameter_types
/ros_gz_bridge/get_parameters
/ros_gz_bridge/get_type_description
/ros_gz_bridge/list_parameters
/ros_gz_bridge/set_parameters
/ros_gz_bridge/set_parameters_atomically
/world/empty/control

```

What am I missing? Are there additional dependencies or message type conversions required for these particular services?

masterccioli, 2025-04-22:
It looks like the solution for this was pushed on March 21st: https://github.com/ros-simulation/simulation_interfaces/pull/1

danicannt, 2025-05-12:
I am having the exact same problem in ROS2 Humble

Instead of launching the parameter bridge via a launch file, I'm attempting to run it directly using the following command:
`ros2 run ros_gz_bridge parameter_bridge /world/empty/create@ros_gz_interfaces/srv/SpawnEntity
`
My goal is to call this service from a ROS 2 Humble node I've developed, in order to spawn objects into the Ignition simulation. Any help or guidance would be greatly appreciated.

closed 2026-03-11 by azeey