gazebosim/ros_gz #716: Uniform launch systems

dvdmc, 2025-03-26:
While doing the tutorial for [Launch Gazebo from ROS 2](https://gazebosim.org/docs/harmonic/ros2_launch_gazebo/#). I noticed the differences between launch files is not completely clear. When checking the [repo files](https://github.com/gazebosim/ros_gz/tree/ros2/ros_gz_sim/launch), there are plenty. While the [tutorial](https://gazebosim.org/docs/harmonic/ros2_launch_gazebo/#launching-with-ros-gz-bridge) seems to imply that using gz_sim & ros_gz_sim only includes the ros_gz_bridge, they are quite different in their usage, and many additional parameters should be specified; the gz_args are not documented in that page...

The question here is: What is the current recommended usage for new users to focus on? Does it make sense to try and unify [gz_sim.launch.py.in](https://github.com/gazebosim/ros_gz/blob/jazzy/ros_gz_sim/launch/gz_sim.launch.py.in) with the rest?

dvdmc, 2025-03-26:
Thanks for the clarification. I will check and refer to #708 from now. Just an additional question: are you aware of any reasons why these launch files cannot be added to `Humble`. We are currently developing on that one and it seems that even when installing the `ros-humble-ros-gzharmonic`, the `Humble` branch is used, which doesn't include these really useful launch files and launch `Actions`

Amronos, 2025-03-26:
> Just an additional question: are you aware of any reasons why these launch files cannot be added to Humble? We are currently developing on that one and it seems that even when installing the [...]

@caguero can you provide info on this?

dvdmc, 2025-03-26:
Thanks! All clear now :) Love to know that you will update nav2 launch files.

Feel free to close when we get info about the possibility of compiling Jazzy or Rolling on Humble.

Amronos, 2025-03-26:
> How are we supposed to run the client?

Here is how you can do it in python:

```python
from launch.actions import ExecuteProcess

gz_gui = ExecuteProcess(cmd=['gz', 'sim', '-g'], output='screen')
```

dvdmc, 2025-03-26:
Thanks! If this will be the standard. Should it be added to the docs?

Amronos, 2025-03-26:
> Thanks! If this will be the standard. Should it be added to the docs?

I suppose it can.

azeey, 2026-03-11:
It would be great to add info about running the GUI to https://gazebosim.org/docs/jetty/ros2_launch_gazebo/

swayanshree07-gif, 2026-11-01:
Hi, 
I would like to work on this and add a section about running the GUI to https://gazebosim.org/docs/jetty/ros2_launch_gazebo/

Before I open a PR on gazebosim/docs, I have a couple of questions:
    
**1. Which versions should get it?** `ros2_launch_gazebo.md` is in `jetty/`, `rotary/`, `harmonic/` and `ionic/`. Should I add it to `jetty` only, or to `rotary` too (and maybe `harmonic`/`ionic`)?
**2. What's the recommended way to start the GUI?** Is `ExecuteProcess` with `gz sim -g` the way you would like the docs to show? Or should the tutorial use a launch option, like including `gz_sim.launch.py` with `gz_args:='-g'`? Earlier in this thread, Amronos suggested this for when the server is started with `gz_server.launch.py` / `ros_gz_sim.launch.py`:
```bash
from launch.actions import ExecuteProcess
gz_gui = ExecuteProcess(cmd=['gz', 'sim', '-g'], output='screen')
```

Thanks!