gazebosim/ros_gz #725: Connecting Gazebo Topics to ROS 2 Topics Across Docker Containers Using ros_gz_bridge

nikisim, 2025-04-10:
Thank you for your ros_gz_bridge tool! I faced interesting case using ros_gz_bridge and ros2 in docker container:

**1. Problem Statement**
I am currently using ros_gz_bridge to interface between ROS 2 and Gazebo. I have a Gazebo simulation running, and I'm attempting to connect to it from a ROS 2 environment inside a Docker container. However, I'm unable to see the Gazebo topics in the ROS 2 environment. I am seeking guidance on how to bridge these topics across the Docker container boundary. Is this functionality supported by ros_gz_bridge?

**2. Current Setup**
Gazebo Simulation : Running on the host machine.
ROS 2 Environment : Running inside a Docker container. I am using --network host for my container
Bridge Tool : ros_gz_bridge is set up on both systems.
**3. Steps Taken**
Started the Gazebo simulation on the host.
Launched the Docker container with ROS 2 setup.
Attempted to use ros_gz_bridge to bridge topics from the host to the container.
**4. Challenges Faced**
Gazebo topics are not visible in the ROS 2 Docker container.
Unclear how to establish a bridge between Gazebo topics and ROS 2 topics when across container boundaries.
**5. Question**
Is there a recommended method or configuration to allow ros_gz_bridge to connect Gazebo topics from the host machine into a ROS 2 environment running inside a Docker container? If so, could you provide guidance or reference documentation on setting up such a configuration?

Thank you for your assistance. I look forward to any suggestions or information you can provide.

Nikita Simonov

gm14, 2025-10-25:
This is pretty old, but in case you or anybody sees this and wants a solution, I solved it by doing 

`export GZ_PARTITION=my_gz_world` in both the host and docker container. The reason you weren't able to see the gz topics is because it was running on a different partition inside the container. The default partition name is "<HOSTNAME>:<USERNAME> where <HOSTNAME> is the hostname of the current machine and <USERNAME> is the name of the user launching the node." So, of course, then gazebo is running on a different partition inside your container.

Not-NeoN-sup, 2026-03-18:
it seems i have done it wrong and gazebo needs to run in desktop while ROS runs in Gazebo I will make changes