gazebosim/ros_gz #726: Should the launch action create component_container_mt instead of just component_container ?

peci1, 2025-04-10:
## Desired behavior
I've learned yesterday that the default `component_container` is single-threaded. And ros_gz uses that in the launch action:

https://github.com/gazebosim/ros_gz/blob/9e2bc76457551eceb57db39f58b73530f0332618/ros_gz_bridge/ros_gz_bridge/actions/ros_gz_bridge.py#L210

Wouldn't it be better to use the multithreaded version? I guess the bridge node can get quite busy at times and running in a multi-threaded container might be desirable.

mjcarroll, 2025-04-11:
I think that all the bridge subscriptions are re-entrant, so this would be a safe choice.

In general, rather than hard-coding an executor, perhaps the better approach would be to provide a parameter that would allow a user to specify the executor (with a sane default).  This would allow for potentially using the EventsExecutor implementation.

peci1, 2025-04-11:
Hmm, allowing to configure EE is an interesting idea. However, that would require changes on the rclcpp_components side, as the executor is hard-coded: https://github.com/ros2/rclcpp/blob/rolling/rclcpp_components/src/component_container_mt.cpp .

An easier path here could now be to just expose the name of the container binary as an attribute of the launch action. Would that make sense? Can we default to the _mt one, or should the single-threaded one be default to keep backwards compatibility?