gazebosim/ros_gz #727: ros_gz_bridge not work for Humble and Harmonic

jinaoao, 2025-04-12:
OS: Ubuntu 22.04
ROS: Humble
Gazebo: Harmonic

I build the ros_gz from source, following the instructions of README.md. But the ros_gz_bridge did not work. The message cannot send/receive through the bridge between the ROS and Gazebo.

andreaceola, 2025-05-15:
Hi @jinaoao ! Do you find any solutions? I have the same problem: ros2 topic list shows the bridged topics, but ros2 topic echo has no output (while gz topic -e shows that the topics are being published)

fzoric8, 2025-11-18:
@rslim97 

If i remember correctly, if you have `humble` and `GZ_VERSION=harmonic` you have to: 
1) Delete previous `gz_sim` installation 
2) Install `gz_sim` using: `sudo apt-get install ros-humble-ros-gzharmonic` 
3) Source everything again 

And it should work.

azeey, 2026-03-11:
The officially supported version of Gazebo for ROS 2 Humble, is Fortress (see https://gazebosim.org/docs/jetty/ros_installation/). If you install `ros-humble-ros-gzharmonic`, you will need to build anything that depends on Gazebo (e.g. `gz_ros2_control`) from source because the one available from binaries was compiled against Fortress. You should also make sure Fortress is not installed to avoid any mixups.

I will go ahead and close this since this does not appear to be a bug in ros_gz.