gazebosim/ros_gz #728: libEGL warning: NEEDS EXTENSION: falling back to kms_swrast

jhaveri-bhavya, 2025-04-12:
Issue: 
Cannot run Gazebo simulation inside Ros2 Jazzy

Steps to reproduce: 
1) In Ubuntu 24.04 terminal source ros2 jazzy
`source /opt/ros/jazzy/setup.bash`

2) run gazebo simulation
`gz sim`

3) Gazebo simulation dialog opens. Open any default loaded environment like "Tugbot in warehouse"

Actual Result:
The selected environment opens but is very very laggy. Cannot zoom in zoom out or do any other tasks
The terminal returns:
```
libEGL warning: MESA-LOADER: egl: failed to open vgem: driver not built!

libEGL warning: NEEDS EXTENSION: falling back to kms_swrast
libEGL warning: MESA-LOADER: egl: failed to open vgem: driver not built!

libEGL warning: NEEDS EXTENSION: falling back to kms_swrast

```
Expected Result:
The simulation environment should run smoothly without any issues. 

Device Specs: 
Windows 11 Home HP Victus Gaming Laptop 15 
13th Gen Intel i5 -13420 
NVIDIA GeForce RTX 3050 6GB 
Laptop GPU Driver version:  572.83

jhaveri-bhavya, 2025-04-16:
@azeey Could you please assist me on this further?

azeey, 2026-03-11:
Thanks for the issue, but this is a type of question that is best asked on https://robotics.stackexchange.com/. I encourage you to post your question there and if you find that there's a specific bug or feature request, come back and create a new issue here. In the mean time, I'll go ahead and close this.

closed 2026-03-11 by azeey