gazebosim/ros_gz #732: Add support for ros-simulation/simulation_interfaces 1.0.0 in Ionic

scpeters, 2025-04-24:
## Desired behavior

Standard ROS interfaces have been defined for interacting with simulation in the [ros-simulation/simulation_interfaces](https://github.com/ros-simulation/simulation_interfaces) package, which has just had its [1.0.0 release](https://github.com/ros-simulation/simulation_interfaces/tree/1.0.0). We should consider how to support these interfaces in Gazebo. Since the upcoming ROS 2 Kilted release will [use Gazebo Ionic](https://www.ros.org/reps/rep-2000.html#kilted-kaiju-may-2025-november-2026), we should add support for Ionic if possible.

## Alternatives considered

## Implementation suggestion

Options:

* Use [ros_gz_bridge](https://github.com/gazebosim/ros_gz/tree/ros2/ros_gz_bridge), mapping to existing protos used in Gazebo and adding/duplicating missing protos as needed.
* Write code to start ROS services using simulation_interfaces and interface with Gazebo's current interfaces.

## Additional context

scpeters, 2025-04-28:
We discussed this during todays Gazebo PMC meeting.

* There is support for implementing these interfaces in Gazebo to allow users to more easily switch between simulators and to allow simulators to focus on what they are best at.
* There was a concern raised about not wanting to negatively impact existing code that doesn't use these interfaces and a suggestion to implement the interfaces in a separate repository (either here in `ros_gz` or another one) rather than pushing changes up to the native gazebo packages.
* The next question is which interfaces to implement first. We shouldn't feel obligated to implement them all but will make a good faith effort.

Please feel free to continue discussion here.

Cavalletta98, 2025-08-29:
Hi @azeey are going to support also Gazebo Harmonic? If not, how we can leverage these standard interfaces into Gazebo Harmonic?

azeey, 2025-10-16:
https://github.com/gazebosim/ros_gz/pull/790 has been merged and released into Rolling which is currently paired with Gazebo Jetty. The plan is to backport it to Kilted and Jazzy. Humble and Iron will not be supported for now as they would require additional work to backport the `gzserver` node.

The following upstream (Gazebo) PRs will first need to be backported all the way to Harmonic, which is the Gazebo version paired with Jazzy:
- https://github.com/gazebosim/gz-transport/pull/671
- https://github.com/gazebosim/gz-sim/pull/2996 followed by https://github.com/gazebosim/gz-sim/pull/2999
- https://github.com/gazebosim/gz-sim/pull/3002
- https://github.com/gazebosim/gz-sim/pull/3005

Cavalletta98, 2025-10-17:
Thank you @azeey . When will be official release the new version of Gazebo Harmonic with the simulation interfaces?