gazebosim/ros_gz #770: ros-humble-ros-gzharmonic depends on python3-matplotlib 3.5 (APT), blocking use of newer versions (e.g. 3.10) in venv

PhPernet, 2025-06-12:
## Description

The Debian package `ros-humble-ros-gzharmonic` declares a dependency on `python3-matplotlib`, which on Ubuntu 22.04 resolves to version **3.5.1**. This causes version conflicts for projects that use a more recent `matplotlib` version (e.g. 3.10.x) in a Python virtual environment.

Even inside a venv, Python sees both the local and system packages, which results in `matplotlib` import errors or inconsistent behavior.

In particular, I am unable to use `Axes3D` for 3D plotting. The following warning is shown:

```
/home/ppernet/ac_venv/lib/python3.10/site-packages/matplotlib/projections/__init__.py:63: UserWarning: Unable to import Axes3D. This may be due to multiple versions of Matplotlib being installed (e.g. as a system package and as a pip package). As a result, the 3D projection is not available.
```
If I remove `python3-matplotlib` using apt, which also removes `ros-humble-ros-gzharmonic`, the issue disappears and 3D plotting works correctly.

## System Info

- **OS**: Ubuntu 22.04  
- **ROS 2 Distro**: Humble  
- **ros_gz version**: installed via `apt` (`ros-humble-ros-gzharmonic`)  
- **Python**: 3.10  
- **matplotlib (system)**: 3.5.1 from `python3-matplotlib` (APT)  
- **matplotlib (venv)**: 3.10.3 installed via `pip`

## Expected Behavior

- Either: `ros-humble-ros-gzharmonic` should not require `python3-matplotlib` unless it truly needs it  
- Or: the dependency should be relaxed, made optional, or documented  
- Ideally: allow users to use a newer `matplotlib` version from a venv without causing conflicts

## Notes

I couldn't find any direct import or usage of `matplotlib` in the `ros_gz` source code.  
If the dependency is legacy or coming from a subcomponent, perhaps it could be removed or made optional?

Related upstream issue: https://github.com/matplotlib/matplotlib/issues/27342

sickopickle, 2025-07-11:
@PhPernet have you found a solution/workaround yet?

PhPernet, 2025-07-14:
@sickopickle unfortunately not